semiotproject/semiot-device-prototype #19: Configuration interface

KMax, 2016-05-21:
The SDK should provide a way to configure a device through the CoAP API. The API:

 - `/config` - ...
    - GET request returns the current configuration in JSON-LD format or 404 if it's not configured
    - PUT request accepts a JSON-LD message with a new confgiuration
 - `/config/schema` - GET request return the configuration schema in JSON-LD format.
 - `/config/context` - GET request should return the JSON-LD context

The JSON-LD context is hardcoded by default, but developer should be able to replace it. The default context:
```
{
  "@context": {
    "xsd": "http://www.w3.org/2001/XMLSchema#",
    "rdfs": "http://www.w3.org/2000/01/rdf-schema#",
    "@vocab": "/config/schema"
  }
}
```

Below is an example of the configuration schema presented. The name of the each configuration field should be an URI, but we don't want to force the developer to bother about prefixes, so [`@vocab`](https://www.w3.org/TR/json-ld/#default-vocabulary) is used to say that any key without a prefix is belongs to the configuration schema vocabulary:
```
{
  "@context": "/config/context",
  "wifi-name": {
    "@type": "xsd:string",
    "rdfs:label": "Wi-Fi Network Name"
  },
  "wifi-password": {
    "@type": "xsd:string",
    "rdfs:label": "Wi-Fi Password"
  }
}
```

Below is an example of the configuration:
```
{
  "@context": "/config/context",
  "wifi-name": "ISST",
  "wifi-password": "blahbalh"
}
```

a-andreyev, 2016-09-15:
Basic auth option number: `40`
all the sensitive changes should contain basic auth option value.

Current basic auth option value:
sha1 sum from `res-login` + `res-pass` string from `/config` resource

Public avaliable resources from the public network are:
`/.well-known/core`
`/doc`

a-andreyev, 2016-09-15:
`/doc`:
{
  "@context": "/doc/context",
  "label": "New device label",
  "id": "e9704afee1cf69a25b38fbc927ed7da9f0c23ac2"
}

a-andreyev, 2016-10-25:
updated and separated to: https://github.com/semiotproject/semiot-device-prototype/wiki/Configuration-interface

closed 2016-11-16 by a-andreyev