semiotproject/semiot-device-prototype #22: Датчики для мед. центра

a-andreyev, 2016-10-05:
# для медицинского центра -- 10 устройств dht от батарейки:

+ минимальные требования: 1 месяц от батарейки при опросе датчика раз в 30 минут
+ утилита для быстрого конфигурирования множества устройств
+ совместно с олимпом или другим фаблабом посмотреть в сторону печатаемых на 3d-принтере корпусов
+ печатная плата? для ~1000 устройств на будущее

nikitachrs, 2016-10-06:
Для сборки одного устройства предлагаю взять недостающие компоненты в магазине [tec](tec.org.ru) и [roboparts](roboparts.ru):

- расширительная плата 	45 руб
- ams1117mini 			85 руб
- батарейный отсек 		25 руб
- кнопки 					10 руб
- корпус					45/75 руб
- транзистор				3 руб
- мгтф						4 руб
- dht22

По теоретическим расчётам, требуется корпус с размерами **90х56х30** или **112х79х32**. Стоимость закупки деталей для одного устройства составляет **240** руб. без включения стоимости самого модуля esp и dht.

a-andreyev, 2016-10-06:
dht22 тоже придется взять, сейчас у нас только несколько 11-ых

closed 2016-11-16 by a-andreyev